doudz/zigate #193: Event is not JSON serializable: <Event zigate.attribute_updated[L] ...>

csacre, 2020-07-07:
Thank you for your work first ;-)

Since latest HassIO Upgrade I notice that AppDaemon4 is disconnected every 5 or 6 min from Hassio.
Looking further in the main log, I notice that apparently there are some "events" from Zigate that cannot be serialized in JSON. This seems to come from all my zigbees.

Here are 3 of these errors

`
2020-07-07 12:22:09 WARNING (Recorder) [homeassistant.components.recorder] Event is not JSON serializable: <Event zigate.attribute_updated[L]: endpoint=1, cluster=0, addr=b668, attribute=65281, data=0121b30b0421a81305210700062402000000006429070965215c180a210000, name=xiaomi, type=dict, value=1=2995, 4=5032, 5=7, 6=b'\x02\x00\x00\x00\x00', 100=2311, 101=6236, 10=0, ieee=00158d0003a28772, device_type=lumi.sensor_ht, entity_id=zigate.00158d0003a28772>
2020-07-07 12:22:09 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.1667796816] Unable to serialize to JSON. Bad data found at $.event.data.value.6=b'\x02\x00\x00\x00\x00'(<class 'bytes'>

2020-07-07 12:28:02 WARNING (Recorder) [homeassistant.components.recorder] Event is not JSON serializable: <Event zigate.attribute_updated[L]: endpoint=1, cluster=0, addr=e843, attribute=65281, data=0121a90b0421a81305211100062402000000006429b408652138170a2114ba, name=xiaomi, type=dict, value=1=2985, 4=5032, 5=17, 6=b'\x02\x00\x00\x00\x00', 100=2228, 101=5944, 10=47636, ieee=00158d0003a27993, device_type=lumi.sensor_ht, entity_id=zigate.00158d0003a27993>
2020-07-07 12:28:02 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.1690188336] Unable to serialize to JSON. Bad data found at $.event.data.value.6=b'\x02\x00\x00\x00\x00'(<class 'bytes'>

2020-07-07 12:34:34 WARNING (Recorder) [homeassistant.components.recorder] Event is not JSON serializable: <Event zigate.attribute_updated[L]: endpoint=1, cluster=0, addr=a5db, attribute=65281, data=0121a90b0421a813052115000624010000000064293f09652188180a2114ba, name=xiaomi, type=dict, value=1=2985, 4=5032, 5=21, 6=b'\x01\x00\x00\x00\x00', 100=2367, 101=6280, 10=47636, ieee=00158d0003590e7a, device_type=lumi.sensor_ht, entity_id=zigate.00158d0003590e7a>
2020-07-07 12:34:34 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.1669093520] Unable to serialize to JSON. Bad data found at $.event.data.value.6=b'\x01\x00\x00\x00\x00'(<class 'bytes'>
`

csacre, 2020-07-12:
Extract with Trace

2020-07-11   16:58:55 DEBUG (ZiGate-Listen) [zigate] Raw packet received,   b'\x01\x81\x02\x12\x02\x10,\x88\x8f\x19k\x02\x11\x02\x10\x02\x10\xff\x02\x11\x02\x10B\x02\x10\x1f\x02\x11!\x9f\x02\x1b\x02\x14!\xa8\x13\x02\x15!\x02\x1d\x02\x10\x02\x16$\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10d)\x93\x02\x18e!E\x18\x02\x1a!5\xb5?\x03'
--
2020-07-11 16:58:55 DEBUG (ZiGate-Event   Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Received response 0x8102:   b'8f196b010000ff010042001f01219f0b0421a81305210d000624020000000064299308652145180a2135b5'
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:143,   addr:196b, endpoint:1, cluster:0, attribute:65281, status:0, data_type:66,   size:31, data:01219f0b0421a81305210d000624020000000064299308652145180a2135b5,   lqi:63
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Acquire Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Release Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Acquire Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Release Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Acquire Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Release Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Handle special xiaomi attribute {'endpoint': 1, 'cluster': 0,   'addr': '196b', 'attribute': 65281, 'data':   '01219f0b0421a81305210d000624020000000064299308652145180a2135b5', 'name':   'xiaomi', 'type': <class 'dict'>, 'value': {1: 2975, 4: 5032, 5: 13, 6:   b'\x02\x00\x00\x00\x00', 100: 2195, 101: 6213, 10: 46389}}
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Acquire Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Release Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Acquire Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Release Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Acquire Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Release Lock on device LUMI lumi.sensor_ht (196b)   00158d00034f8c03
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Dispatch ZIGATE_ATTRIBUTE_UPDATED
2020-07-11 16:58:55 DEBUG (ZiGate-Decode   data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2020-07-11 16:58:55 WARNING   (Recorder) [homeassistant.components.recorder] Event is not JSON   serializable: <Event zigate.attribute_updated[L]: endpoint=1, cluster=0,   addr=196b, attribute=65281,   data=01219f0b0421a81305210d000624020000000064299308652145180a2135b5,   name=xiaomi, type=dict, value=1=2975, 4=5032, 5=13,   6=b'\x02\x00\x00\x00\x00', 100=2195, 101=6213, 10=46389,   ieee=00158d00034f8c03, device_type=lumi.sensor_ht,   entity_id=zigate.00158d00034f8c03>
2020-07-11 16:58:55 ERROR (MainThread)   [homeassistant.components.websocket_api.http.connection.1670276688] Unable to   serialize to JSON. Bad data found at   $.event.data.value.6=b'\x02\x00\x00\x00\x00'(<class 'bytes'>

csacre, 2020-07-14:
Problem is coming when updating XIAOMI specific attribute 65281 as the "value" is not serializable in JSON as it contains json attributes as numeric value and not as string.

 'value': {1: 2975, 4: 5032, 5: 13, 6: b'\x02\x00\x00\x00\x00', 100: 2195, 101: 6213, 10: 46389}

To fix this temporarily, I have changed my copy of  __init__.PY

`    def attribute_updated(**kwargs):
        device = kwargs['device']
        ieee = device.ieee
        attribute = kwargs['attribute']
        _LOGGER.debug('Update attribute for device {} {}'.format(device,
                                                                 attribute))
        entity = hass.data[DATA_ZIGATE_DEVICES].get(ieee)
        event_data = attribute.copy()
        if type(event_data.get('type')) == type:
            event_data['type'] = event_data['type'].__name__
        event_data['ieee'] = device.ieee
        event_data['addr'] = device.addr
        event_data['device_type'] = device.get_property_value('type')
        if entity:
            event_data['entity_id'] = entity.entity_id
        **#CS temp fix
        if str(event_data['attribute']) != "65281":
            hass.bus.fire('zigate.attribute_updated', event_data)
        else:
            _LOGGER.warning('ZIGATE CS SKIP ATTRIBUTE 65281')**
`

doudz, 2020-07-15:
it should be fixed in 0.40.1

closed 2020-07-15 by doudz

csacre, 2020-07-15:
It works! Thank you!