doudz/zigate #194: How to join an existing network ?

dusscloud, 2020-08-11:
It seems that by default the python library cset-up the zigate as a coordinator and creates its own network.
But in my cas I want my Zigate to join a existing network, as a simple router.
How can I use the izgate python library to do that?

pipiche38, 2020-08-11:
https://github.com/badzz/Zigate-Router

dusscloud, 2020-08-11:
> 
> https://github.com/badzz/Zigate-Router
> 

Hmmm I miss some guidelines here.

pipiche38, 2020-08-11:
The firmware are available there: https://github.com/badzz/Zigate-Router/tree/master/Router/Build

Then you have to use the same procedure as the standard firmware to flash the router firmware on the ZiGate.

dusscloud, 2020-08-12:
Ok, thanks.
- which one of the binary has to be flashed ?
- Then what are the Python command to be used in order to join an existing network ?