doudz/zigate #201: Supprimer toutes les données pour migration Homeassistante --> ZHA

MichaelBitard, 2020-11-20:
Je suis en train de migrer de https://github.com/doudz/homeassistant-zigate vers la version officielle zha.

J'ai une zigate usb en version 3.1d.
Quand je tente de pairer mes devices avec le nouveau système je me retrouve avec des logs étranges du style :

`2020-11-20 20:25:15 INFO (MainThread) [zigpy.application] Device 0x0000 (00:xx:xx:xx:xx:xx:xx) left the network`

Est-ce parce que le device est enregistré sur la zigate ? Si oui, comment faire pour tout supprimer ?

J'ai tenté : 

```python
>>> import zigate
>>> z = zigate.connect(port=None)
z = z.erase_persistent()
False
>>> Unknown response 0x0302
```
Mais cette unknown response me semble une erreur

doudz, 2020-11-20:
Ce message 0x0302 indique simplement que le pdm a été initialisé. La mémoire est donc vide

MichaelBitard, 2020-11-21:
Merci, en effet, c'est étrange car les éléments étaient toujours là. Mais au bout d'un moment en effet ça c'est bien réinitialisé.

closed 2020-11-21 by MichaelBitard