doudz/zigate #202: Zigate2zigpy utilisable ?

Bredin76, 2020-12-02:
Bonjour,

Je viens d'installer ZHA sur un HASS.IO 0118.4 en partant d'un custom component ZIGATE . Maintenant je cherche à migrer mes équipements et comme j'en ai un certain nombre - la migration par le script zigate2zigpy.py m’intéresse au plus haut. 
Le script n'est pas documenté et je rencontre une anomalie à l’exécution du type
```
pi@raspberrypi:/srv/docker/hassio/homeassistant $ sudo python3 zigate2zigpy.py
Import device  00:15:8d:00:03:46:5e:29
Traceback (most recent call last):
  File "zigate2zigpy.py", line 78, in <module>
    data = unhexlify(data)
binascii.Error: Odd-length string
```
Est il possible d'avoir un conseil d'usage là dessus ?

doudz, 2020-12-02:
Le script n'est pas utilisable en l'état, les bibliothèques ayant évoluées depuis.
Perso je viens de terminer la migration de 21 appareils vers zha, je l'ai fait en refaisant l'appairage des différents équipements.
Dans tous les cas, zha n'utilise pas la même notation pour les  entity donc il sera nécessaire de refaire les scripts, automations, dashboard, etc

Bredin76, 2020-12-02:
Ce réapparaige a t-il nécessité l'effacement de la mémoire de PiZigate ? En effet - j'ai essayé ce matin de désapparairer un Aqara température ce matin et de le réappairer - il ne s'est pas réappairé avec ZHA bien que la let clignote en mode permit. 

Le script m'intéresse quand même car j'ai des ampoules OSRAM juste atroce à réappairer + des HUE sans bridge donc galère. As tu un guide même succinct quelque part qui me permettrait de créer la db ? même à la main au moins pour ces équipements casse pied ?