doudz/zigate #203: installation zigate

sebastienbonichon, 2020-12-02:
Bonjour,

mon jeedom V4 tourne sur une vm proxmox sur un mini pc et je n'arrive pas a installer ma zigate usb. Je dois avouer que je ne sais plus a qui m'adresser et je vois que vous paraissez maitriser cette clé.

pourriez vous jeter un œil sur ce post où j'explique tout : https://community.jeedom.com/t/comment-installer-une-zigate-usb-sur-un-mini-pc/44469

peut être, également, pouvez vous m'aiguiller vers quelqu'un qui pourrait me renseigner

cdlt

seb

doudz, 2020-12-02:
je ne connais pas proxmox mais la conf du port usb semble correct.
Il faudrait mettre le log démon en mode debug pour voir ce qu'il se passe au démarrage

doudz, 2020-12-02:
et pour info, cet issue aurait plutôt sa place ici : https://github.com/jeedom-zigate/jeedom-plugin-zigate/issues

sebastienbonichon, 2020-12-02:
le log abeille.config.log me donne

[2020-12-02 20:20:00] Démarrage de 'checkTTY.sh'
Vérifications du port '/dev/ttyUSB2'
= ERREUR: Le port est utilisé par le process '877'.
= Il doit être libéré et n'être utilisé QUE par le plugin Abeille pour permettre le dialogue avec la Zigate.
= Details du process 877:
= PPid=528, cmd='/usr/sbin/apache2 -k start'

je n'ai acces qu'a l'usb 2 alors que proxmox me "parle" d'usb3
a savoir que j'ai sur cette machine déjà une clé z wave, une rfx et une téléinfo. lorsque je fais la meme manip sur une vm avec un jeedom vierge : ca marche .....

doudz, 2020-12-02:
ah je n'avais pas noté que c'était le plugin Abeille... là je ne peux rien faire

closed 2020-12-02 by doudz

sebastienbonichon, 2020-12-02:
oups, oui mais je n'utilise pas le plugin zigate ......

ah...... dommage, merci de m'avoir répondu