doudz/zigate #207: No actions on Sunricher Micro Smart Dimmer

ngirardin, 2021-01-17:
Hi,

First, thank you for maintaining this library, it makes playing with the zigate really fun :)

I've just bought a [Sunricher Micro Smart Dimmer](https://www.amazon.fr/gp/product/B087G1H2DL/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1), that is reporting some data about its consumption, but no actions:

```
{
  'attribute': 0,
  'data': '\x00\x00\x00\x10\x19~',
  'name': 'current_delivered',
  'type': <class 'float'>, 'value': '\x00\x00\x00\x10\x19~'
}, {
  'attribute': 4,
  'data': 'Sunricher',
  'name': 'manufacturer',
   'value': 'Sunricher'
}, {
  'attribute': 5,
  'data': 'Micro Smart Dimmer',
  'name': 'type',
  'value': 'Micro Smart Dimmer',
  'type': <class 'str'>
}
```

I'm only seeing the `0x702` cluster, but the doc states that it's supposed to support a bunch of others:
```{1: {'clusters': {1794: Cluster 0x0702 Metering}, 'profile': 0, 'device': 0, 'in_clusters': [], 'out_clusters': []}}```

[image]

doudz, 2021-01-17:
it looks like the module is not correctly discovered, in_clusters and out_clusters are not supposed to be empty and device or profile should not be 0.
did you try to call discover again ?
if discover doesn't help and since you have the doc we could create a template to add missing informations

ngirardin, 2021-01-23:
Hi doudz,

I made a factory reset, launched `z.permit_join`, and re-paired the dimmer, but this time I'm even getting less information.

Here are some relevant entries from the logs:

```
DEBUG:zigate:Check Need discovery Device  (18ae) 60a423fffec89cf1
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 18ae
DEBUG:zigate:Need discovery : no type
DEBUG:zigate:Need discovery : no endpoints
DEBUG:zigate:Dispatch ZIGATE_DEVICE_NEED_DISCOVERY
DEBUG:zigate:Auto saving /home/freebox/.zigate.json
DEBUG:zigate:Saving persistent file
DEBUG:zigate:Acquire Lock to save persistent file
DEBUG:zigate:Release Lock of persistent file
DEBUG:zigate:REQUEST : 0x0010
DEBUG:zigate:Msg to send b'0010000010'
DEBUG:zigate:Encoded Msg to send b'01021010021002101003'
DEBUG:zigate:Waiting for status message for command 0x0010
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\x97\x02\x10\x02\x10\x02\x10\x10\x02\x10\x02\x10\x02\x10\x03\x01\x80\x10\x02\x10\x02\x15\x88\x02\x10\x02\x13\x02\x13\x1d\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000100000'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'\x00\x00', lqi:0
DEBUG:zigate:Received response 0x8010: b'0003031d'
DEBUG:zigate:RESPONSE 0x8010 - Version list : major:3, installer:31d, lqi:0, version:3.1d
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0010:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'\x00\x00', lqi:0
Got 1 devices
DEBUG:zigate:Check Need discovery Device  (18ae) 60a423fffec89cf1
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 18ae
DEBUG:zigate:Need discovery : no type
DEBUG:zigate:Need discovery : no endpoints
Device addr: 18ae, name: ,  genericType: , get_type: None, lqi: 100%, battery: 100%
  - device.endpoints: {}
  - device.attributes: []
DEBUG:zigate:Check Need discovery Device  (18ae) 60a423fffec89cf1
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 18ae
DEBUG:zigate:Need discovery : no type
DEBUG:zigate:Need discovery : no endpoints
  - device.need_discovery: True
```

I always have a lot of trouble pairing devices to my zigate: for a few hour I'm only seeing partial information, and it takes several hours to see the device endpoints, and some are missing.

This is the info that I'm getting after pairing my Xiaomi Aqara temp sensor:

```
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 432b
Device addr: 432b, name: ,  genericType: , get_type: None, lqi: 73%, battery: 100%
  - device.endpoints: {}
  - device.attributes: []
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 432b
  - device.need_discovery: True
```

I've run:
```
for device in z.devices:
  device.discover_device()
```
but even after a few minutes the devices still have `need_discovery: True` after that, and no endpoints.

- what is `discover_device` supposed to do?
- is it normal that the discovery is taking several hours and that I'm missing some endpoints?

Thanks for your help

doudz, 2021-01-25:
The log is incomplete, could provide the complete log file ?
Because when it says "DEBUG:zigate:Need discovery : no endpoints" after it should request for active endpoints

ngirardin, 2021-01-25:
Thanks for the quick reply.

Here are the complete logs:
```
DEBUG:zigate:Try loading persistent file
DEBUG:zigate:Trying to load /home/freebox/.zigate.json
DEBUG:zigate:Load neighbours cache: []
DEBUG:zigate:Load success
INFO:zigate:Searching ZiGate port
INFO:zigate:ZiGate found at /dev/ttyUSB0
DEBUG:zigate:REQUEST : 0x0018 b'\x01'
DEBUG:zigate:Msg to send b'001800011801'
DEBUG:zigate:Encoded Msg to send b'010210180210021118021103'
DEBUG:zigate:Waiting for status message for command 0x0018
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\x9f\x02\x10\x02\x10\x02\x10\x18\x02\x10\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000180000'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:24, error:b'\x00\x00', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0018:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:24, error:b'\x00\x00', lqi:0
DEBUG:zigate:REQUEST : 0x0010 
DEBUG:zigate:Msg to send b'0010000010'
DEBUG:zigate:Encoded Msg to send b'01021010021002101003'
DEBUG:zigate:Waiting for status message for command 0x0010
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\x97\x02\x10\x02\x10\x02\x10\x10\x02\x10\x02\x10\x02\x10\x03\x01\x80\x10\x02\x10\x02\x15\x88\x02\x10\x02\x13\x02\x13\x1d\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000100000'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'\x00\x00', lqi:0
DEBUG:zigate:Received response 0x8010: b'0003031d'
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0010:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'\x00\x00', lqi:0
DEBUG:zigate:Waiting for message 0x8010
DEBUG:zigate:RESPONSE 0x8010 - Version list : major:3, installer:31d, lqi:0, version:3.1d
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Stop waiting, got message 0x8010
DEBUG:zigate:REQUEST : 0x0021 b'\x07\x18\xc8\x00'
DEBUG:zigate:Msg to send b'00210004f20718c800'
DEBUG:zigate:Encoded Msg to send b'0102102102100214f2021718c8021003'
DEBUG:zigate:Waiting for status message for command 0x0021
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\xa6\x02\x10\x02\x10\x02\x10!\x02\x10\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000210000'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:33, error:b'\x00\x00', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0021:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:33, error:b'\x00\x00', lqi:0
DEBUG:zigate:REQUEST : 0x0023 b'\x00'
DEBUG:zigate:Msg to send b'002300012200'
DEBUG:zigate:Encoded Msg to send b'010210230210021122021003'
DEBUG:zigate:Waiting for status message for command 0x0023
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\xa4\x02\x10\x02\x10\x02\x10#\x02\x10\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000230000'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:35, error:b'\x00\x00', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0023:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:35, error:b'\x00\x00', lqi:0
DEBUG:zigate:Check network state
DEBUG:zigate:REQUEST : 0x0009 
DEBUG:zigate:Msg to send b'0009000009'
DEBUG:zigate:Encoded Msg to send b'010210021902100210021903'
DEBUG:zigate:Waiting for status message for command 0x0009
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\x8e\x02\x10\x02\x10\x02\x10\x02\x19\x02\x10\x02\x10\x02\x10\x03\x01\x80\x02\x19\x02\x10\x16\x13\x02\x10\x02\x10\x02\x10\x15\x8d\x02\x10\x02\x13\xef\x8e\x02\x18\xc75h\xa7\xe0*4\xf9\x95\xda\x02\x1b\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000090000'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:9, error:b'\x00\x00', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Received response 0x8009: b'000000158d0003ef8e08c73568a7e02a34f995da0b'
DEBUG:zigate:RESPONSE 0x8009 - Network state response : addr:0000, ieee:00158d0003ef8e08, panid:50997, extended_panid:7541242572937205210, channel:11, lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
[... 69 lines cut ...]
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8010: b'0003031d'
DEBUG:zigate:RESPONSE 0x8010 - Version list : major:3, installer:31d, lqi:0, version:3.1d
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0010:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'\x00\x00', lqi:0
Got 2 devices
Device addr: 18ae, name: ,  genericType: , get_type: None, lqi: 100%, battery: 100%
  - device.endpoints: {}
  - device.attributes: []
DEBUG:zigate:Check Need discovery Device  (18ae) 60a423fffec89cf1
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 18ae
DEBUG:zigate:Need discovery : no type
DEBUG:zigate:Need discovery : no endpoints
[... cut ...]
-----------------------------------------------------------------
Device addr: 432b, name: ,  genericType: , get_type: None, lqi: 67%, battery: 100%
  - device.endpoints: {}
  - device.attributes: []
DEBUG:zigate:Check Need discovery Device  (432b) 00158d000588e26c
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 432b
DEBUG:zigate:Need discovery : no type
DEBUG:zigate:Need discovery : no endpoints
  - device.need_discovery: True
-----------------------------------------------------------------
```

doudz, 2021-01-25:
Thanks, could you also provide the content of zigate.json ?

doudz, 2021-01-25:
Supposing z is the zigate instance and abcd the address, could try the following code ? :
`z.discover_device('abcd', force=True)
`

ngirardin, 2021-01-26:
Here is the zigate.json:

```
{
    "devices": [
        {
            "addr": "18ae",
            "discovery": "",
            "endpoints": [],
            "generictype": "",
            "info": {
                "addr": "18ae",
                "id": 0,
                "ieee": "60a423fffec89cf1",
                "lqi": 255,
                "power_type": 1
            },
            "name": ""
        },
        {
            "addr": "432b",
            "discovery": "",
            "endpoints": [],
            "generictype": "",
            "info": {
                "addr": "432b",
                "id": 0,
                "ieee": "00158d000588e26c",
                "lqi": 172,
                "power_type": 0
            },
            "name": ""
        }
    ],
    "groups": {},
    "led": true,
    "neighbours_table": [],
    "scenes": {}
}
```

and here are the logs after running
```
z.discover_device('18ae', force=True)
z.discover_device('432b', force=True)
```

```
DEBUG:zigate:Try loading persistent file
DEBUG:zigate:Trying to load /home/freebox/.zigate.json
DEBUG:zigate:Load neighbours cache: []
DEBUG:zigate:Load success
INFO:zigate:Searching ZiGate port
INFO:zigate:ZiGate found at /dev/ttyUSB0
DEBUG:zigate:REQUEST : 0x0018 b'\x01'
DEBUG:zigate:Msg to send b'001800011801'
DEBUG:zigate:Encoded Msg to send b'010210180210021118021103'
DEBUG:zigate:Waiting for status message for command 0x0018
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\x9f\x02\x10\x02\x10\x02\x10\x18\x02\x10\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000180000'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:24, error:b'\x00\x00', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0018:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:24, error:b'\x00\x00', lqi:0
DEBUG:zigate:REQUEST : 0x0010 
DEBUG:zigate:Msg to send b'0010000010'
DEBUG:zigate:Encoded Msg to send b'01021010021002101003'
DEBUG:zigate:Waiting for status message for command 0x0010
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\x97\x02\x10\x02\x10\x02\x10\x10\x02\x10\x02\x10\x02\x10\x03\x01\x80\x10\x02\x10\x02\x15\x88\x02\x10\x02\x13\x02\x13\x1d\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
[... 137 lines cut ...]
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'002700420207'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:39, packet_type:66, error:b'\x02\x07', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0042:RESPONSE 0x8000 - Status response : status:0, sequence:39, packet_type:66, error:b'\x02\x07', lqi:0
```

I've tried to run `discover_device` several time, but the result is still the same, I still don't have any info on any of my devices:
```
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 18ae
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 432b
Got 2 devices
Device addr: 18ae, name: ,  genericType: , get_type: None, lqi: 100%, battery: 100%
  - device.endpoints: {}
  - device.attributes: []
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 18ae
  - device.need_discovery: True
-----------------------------------------------------------------
Device addr: 432b, name: ,  genericType: , get_type: None, lqi: 68%, battery: 100%
  - device.endpoints: {}
  - device.attributes: []
WARNING:zigate:Neither type (modelIdentifier) nor manufacturer_code for device 432b
  - device.need_discovery: True
-----------------------------------------------------------------
```

doudz, 2021-01-26:
Did you wait enough ? Because after sending a command 0x0042 the device should reply with response 0x8042

ngirardin, 2021-01-26:
I've tried this way:
```python
print('------ SENDING DISCOVERY -----')
z.discover_device('18ae', force=True)
z.discover_device('432b', force=True)

while True:
    print('.', end='')
    time.sleep(1)
```

and got this result (and no message after that):
```
------ SENDING DISCOVERY -----
DEBUG:zigate:discover_device 18ae
DEBUG:zigate:no mac_capability
DEBUG:zigate:REQUEST : 0x0042 18ae
DEBUG:zigate:Msg to send b'00420002f618ae'
DEBUG:zigate:Encoded Msg to send b'0102104202100212f618ae03'
DEBUG:zigate:Waiting for status message for command 0x0042
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\xe7\x02\x10*\x02\x10B\x02\x12\x02\x1a\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'002a0042020a'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:42, packet_type:66, error:b'\x02\n', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0042:RESPONSE 0x8000 - Status response : status:0, sequence:42, packet_type:66, error:b'\x02\n', lqi:0
DEBUG:zigate:discover_device 432b
DEBUG:zigate:no mac_capability
DEBUG:zigate:REQUEST : 0x0042 432b
DEBUG:zigate:Msg to send b'0042000228432b'
DEBUG:zigate:Encoded Msg to send b'010210420210021228432b03'
DEBUG:zigate:Waiting for status message for command 0x0042
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\xe7\x02\x10+\x02\x10B\x02\x12\x02\x1b\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'002b0042020b'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:43, packet_type:66, error:b'\x02\x0b', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0042:RESPONSE 0x8000 - Status response : status:0, sequence:43, packet_type:66, error:b'\x02\x0b', lqi:0
.......DEBUG:zigate:Raw packet received, b'\x01\x80\x11\x02\x10\x02\x18\x80\xa7\x18\xae\x02\x10\x02\x10\x02\x12\x02\x1a\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8011: b'a718ae0000020a'
DEBUG:zigate:RESPONSE 0x8011 - APS_DATA_ACK : status:167, addr:18ae, endpoint:0, cluster:2, additional:b'\n', lqi:0
ERROR:zigate:Device 18ae doesn't receive last command to endpoint 0 cluster 2: 0xa7
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Raw packet received, b'\x01\x80\x11\x02\x10\x02\x18_\xa7C+\x02\x10\x02\x10\x02\x12\x02\x1b\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8011: b'a7432b0000020b'
DEBUG:zigate:RESPONSE 0x8011 - APS_DATA_ACK : status:167, addr:432b, endpoint:0, cluster:2, additional:b'\x0b', lqi:0
ERROR:zigate:Device 432b doesn't receive last command to endpoint 0 cluster 2: 0xa7
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
...DEBUG:zigate:Raw packet received, b'\x01\x87\x02\x11\x02\x10\x02\x15\xe5\xd0\x02\x10\x18\xae\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8701: b'd00018ae'
DEBUG:zigate:RESPONSE 0x8701 - Route Discovery Confirmation : status:208, network_status:0, additional:b'\x18\xae', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
.......................................
```

doudz, 2021-01-26:
could you try to call z.reset() ?

ngirardin, 2021-01-27:
```------ SENDING reset -----
DEBUG:zigate:REQUEST : 0x0011 
DEBUG:zigate:Msg to send b'0011000011'
DEBUG:zigate:Encoded Msg to send b'01021011021002101103'
.DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x17\x96\x02\x10\x02\x10\x02\x10\x11\x02\x10\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'000000110000'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:17, error:b'\x00\x00', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Raw packet received, b'\x01\x02\x13\x02\x12\x02\x10\x02\x12\x02\x13\x02\x10\x02\x10\x03\x01\x80$\x02\x10\x02\x1dQ\x02\x11\x02\x10\x02\x10\x02\x10\x15\x8d\x02\x10\x02\x13\xef\x8e\x02\x18\x02\x1b\x02\x10\x03\x01\x80\x02\x16\x02\x10\x02\x12\x85\x02\x11\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x0302: b'00'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
WARNING:zigate:Unknown response 0x0302
DEBUG:zigate:RESPONSE 0x0000 - Base response : additional:b'\x00', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Received response 0x8024: b'01000000158d0003ef8e080b'
DEBUG:zigate:RESPONSE 0x8024 - Network joined / formed : status:1, lqi:0, addr:0000, ieee:00158d0003ef8e08, channel:11
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8006: b'01'
DEBUG:zigate:RESPONSE 0x8006 - Non “Factory new” Restart : status:1, lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
..............```

I still can see any info on the devices even after forcing discover etc...

I'm already on the latest 3.1d firmware, should I reflash it?

ngirardin, 2021-01-30:
Hi @doudz,

I was able to factory reset my zigate and have my Xiaomi temp sensor and my Sunricher dimmer paired again. The devices needed discovery, but running `discover_device` with `force` finally worked!

Here are the device informations:

```
Device addr: 9c72, name: ,  genericType: , get_type: Micro Smart Dimmer, lqi: 32%, battery: 100%
  - device.endpoints: {1: {'clusters': {1794: Cluster 0x0702 Metering, 0: Cluster 0x0000 General: Basic}, 'profile': 260, 'device': 257, 'in_clusters': [0, 3, 4, 5, 6, 8, 1794, 2820, 2821, 4096], 'out_clusters': [25]}}
    - 1
    - action: 1
  - device.attributes: [{'endpoint': 1, 'cluster': 1794, 'attribute': 0, 'data': '\x00\x00\x00\x19ɚ', 'name': 'current_delivered', 'type': <class 'float'>, 'value': '\x00\x00\x00\x19ɚ'}, {'endpoint': 1, 'cluster': 0, 'attribute': 4, 'data': 'Sunricher', 'name': 'manufacturer', 'value': 'Sunricher'}, {'endpoint': 1, 'cluster': 0, 'attribute': 5, 'data': 'Micro Smart Dimmer', 'name': 'type', 'value': 'Micro Smart Dimmer', 'type': <class 'str'>}]
WARNING:zigate:No response waiting command 0x8030
WARNING:zigate:No response waiting command 0x8030
ERROR:zigate:Command 0x0120 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b'\x01^'
ERROR:zigate:Command 0x0030 failed Failed (ZigBee event codes) 0x80 : b'\x02b'
ERROR:zigate:Command 0x0030 failed Failed (ZigBee event codes) 0x80 : b'\x02k'
ERROR:zigate:Command 0x0120 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b'\x01n'
ERROR:zigate:Command 0x0120 failed E_ZCL_ERR_ZTRANSMIT_FAIL : b'\x01p'
  - device.need_discovery: False
  - attribute:
    - name: current_delivered
    - value: ɚ
  - attribute:
    - name: manufacturer
    - value: Sunricher
  - attribute:
    - name: type
    - value: Micro Smart Dimmer
```

And here is the zigate.json, it looks better than before :)

```
{
            "addr": "9c72",
            "discovery": "",
            "endpoints": [
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "data": "\u0000\u0000\u0000\u0019\u025a",
                                    "name": "current_delivered",
                                    "type": "float",
                                    "value": "\u0000\u0000\u0000\u0019\u025a"
                                }
                            ],
                            "cluster": 1794
                        }
                    ],
                    "device": 257,
                    "endpoint": 1,
                    "in_clusters": [
                        0,
                        3,
                        4,
                        5,
                        6,
                        8,
                        1794,
                        2820,
                        2821,
                        4096
                    ],
                    "out_clusters": [
                        25
                    ],
                    "profile": 260
                }
            ],
            "generictype": "",
            "info": {
                "addr": "9c72",
                "bit_field": "0100000000000001",
                "descriptor_capability": "00000000",
                "id": 40,
                "ieee": "60a423fffec89cf1",
                "last_seen": "2021-01-30 14:06:13",
                "lqi": 77,
                "mac_capability": "10001110",
                "manufacturer_code": "1224",
                "max_buffer": 82,
                "max_rx": 82,
                "max_tx": 82,
                "power_type": 1,
                "server_mask": 11264
            },
            "name": ""
        }
```

Can I help you implement the support? 

Thanks,

ngirardin, 2021-02-12:
I reflashed the zigate and everything paired without any issues in a few seconds. I was finally able to send dim and on/off actions to the dimmer. Thanks for the help!

closed 2021-02-12 by ngirardin