doudz/zigate #210: Erreur de dépendance Zigate

genoue, 2023-12-30:
Bonjour,

Depuis quelques jours, les dépendances du plugin Zigate ne s'installent plus.
Je suis sur :
Jeedom 4.3.21 sur smart
Zigate USB
pluggin zigate 1.7.6

J'ai effectué la procédure précisée dans la doc zigate :
$ pip2 -V
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)
$ pip3 -V
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.5)
$ pip -V
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)
Si ce n’est pas le cas :
$ sudo python3 -m pip uninstall pip
$ sudo apt-get remove python3-pip python-pip
$ sudo apt-get install python3-pip python-pip
$ sudo python2 -m pip uninstall pip
$ sudo apt-get remove python-pip
$ sudo apt-get install python-pip
Si ça ne fonctionne toujours pas vérifier que pip existe:
$ which pip
/usr/bin/pip
Si la commande ne retourne rien c’est que pip est manquant, il faut alors le recréer:
$ sudo cp /usr/bin/pip2 /usr/bin/pip

Or de mon côté, c'est pip 18.1.
Est-cela le problème ? Mais impossible d'installer la version 9 :-(

J'ai également fait la procédure précisée dans le sujet :
https://github.com/doudz/zigate/issues/182
sudo apt-get remove python3-serial
sudo python3 -m pip uninstall pyserial
sudo apt remove python3-serial
sudo python3 -m pip install pyserial

Voici le log zigate_update :
0000|Installation des dépendances
0001|Hit:1 http://security.debian.org/debian-security buster/updates InRelease
0002|Get:2 http://ftp.uk.debian.org/debian buster InRelease [122 kB]
0003|Hit:3 https://deb.nodesource.com/node_18.x nodistro InRelease
0004|Get:4 http://ftp.uk.debian.org/debian buster-updates InRelease [56.6 kB]
0005|Hit:5 https://oph.mdrjr.net/meveric all InRelease
0006|Hit:6 https://oph.mdrjr.net/meveric buster InRelease
0007|Hit:7 http://www.deb-multimedia.org buster InRelease
0008|Fetched 178 kB in 5s (34.6 kB/s)
0009|Reading package lists...
0010|Reading package lists...
0011|Building dependency tree...
0012|Reading state information...
0013|python3 is already the newest version (3.7.3-1).
0014|0 upgraded, 0 newly installed, 0 to remove and 255 not upgraded.
0015|Reading package lists...
0016|Building dependency tree...
0017|Reading state information...
0018|python3-pip is already the newest version (18.1-5).
0019|0 upgraded, 0 newly installed, 0 to remove and 255 not upgraded.
0020|Reading package lists...
0021|Building dependency tree...
0022|Reading state information...
0023|python3-serial is already the newest version (3.4-4).
0024|0 upgraded, 0 newly installed, 0 to remove and 255 not upgraded.
0025|Reading package lists...
0026|Building dependency tree...
0027|Reading state information...
0028|python3-usb is already the newest version (1.0.2-1).
0029|0 upgraded, 0 newly installed, 0 to remove and 255 not upgraded.
0030|Reading package lists...
0031|Building dependency tree...
0032|Reading state information...
0033|python3-requests is already the newest version (2.21.0-1+deb10u1).
0034|0 upgraded, 0 newly installed, 0 to remove and 255 not upgraded.
0035|Reading package lists...
0036|Building dependency tree...
0037|Reading state information...
0038|python3-setuptools is already the newest version (40.8.0-1).
0039|0 upgraded, 0 newly installed, 0 to remove and 255 not upgraded.
0040|Reading package lists...
0041|Building dependency tree...
0042|Reading state information...
0043|python3-rpi.gpio is already the newest version (0.6.5-1).
0044|0 upgraded, 0 newly installed, 0 to remove and 255 not upgraded.
0045|Reading package lists...
0046|Building dependency tree...
0047|Reading state information...
0048|python3-dev is already the newest version (3.7.3-1).
0049|0 upgraded, 0 newly installed, 0 to remove and 255 not upgraded.
0050|Collecting zigate==0.40.11
0051|Using cached https://files.pythonhosted.org/packages/d8/d7/64f620efbe1abcaf59d17c5a4168f86dd2ece1c2993f3a7dd79f08625a14/zigate-0.40.11.tar.gz
0052|Requirement already satisfied: pyserial>=3.2 in /usr/local/lib/python3.7/dist-packages (from zigate==0.40.11) (3.5)
0053|Requirement already satisfied: pydispatcher>=2.0.5 in /usr/local/lib/python3.7/dist-packages (from zigate==0.40.11) (2.0.7)
0054|Requirement already satisfied: pyusb in /usr/lib/python3/dist-packages (from zigate==0.40.11) (1.0.2)
0055|Requirement already satisfied: bottle in /usr/local/lib/python3.7/dist-packages (from zigate==0.40.11) (0.12.25)
0056|Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from zigate==0.40.11) (2.21.0)
0057|Building wheels for collected packages: zigate
0058|Running setup.py bdist_wheel for zigate: started
0059|Running setup.py bdist_wheel for zigate: finished with status 'error'
0060|Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-rfygs3oj/zigate/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('
0061|', '
0062|');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-uhn0tall --python-tag cp37:
0063|/usr/local/lib/python3.7/dist-packages/setuptools/dist.py:745: SetuptoolsDeprecationWarning: Invalid dash-separated options
0064|!!
0065|********************************************************************************
0066|Usage of dash-separated 'description-file' will not be supported in future
0067|versions. Please use the underscore name 'description_file' instead.
0068|This deprecation is overdue, please update your project and remove deprecated
0069|calls to avoid build errors in the future.
0070|See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
0071|********************************************************************************
0072|!!
0073|opt = self.warn_dash_deprecation(opt, section)
0074|/usr/local/lib/python3.7/dist-packages/setuptools/config/setupcfg.py:293: _DeprecatedConfig: Deprecated config in `setup.cfg`
0075|!!
0076|********************************************************************************
0077|The license_file parameter is deprecated, use license_files instead.
0078|This deprecation is overdue, please update your project and remove deprecated
0079|calls to avoid build errors in the future.
0080|See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
0081|********************************************************************************
0082|!!
0083|parsed = self.parsers.get(option_name, lambda x: x)(value)
0084|usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
0085|or: -c --help [cmd1 cmd2 ...]
0086|or: -c --help-commands
0087|or: -c cmd --help
0088|error: invalid command 'bdist_wheel'
0089|----------------------------------------
0090|Failed building wheel for zigate
0091|Running setup.py clean for zigate
0092|Failed to build zigate
0093|Installing collected packages: zigate
0094|Running setup.py install for zigate: started
0095|Running setup.py install for zigate: finished with status 'done'
0096|Successfully installed zigate
0097|usermod: group 'gpio' does not exist
0098|Installation des dépendances terminée !
0099|Vérification de la version installée
0100|Traceback (most recent call last):
0101|File "", line 1, in 
0102|ModuleNotFoundError: No module named 'zigate'

merci par avance de votre aide
Stéphane

genoue, 2024-01-03:
Bonjour,
Autre info, je suis passé, il y a deux mois à jeedom v4.
Este le problème ? car je vois dans le log que module zigate n'est pas trouvé.
Merci pour vos retours

doudz, 2024-01-03:
Bonjour, je suis désolé ce plugin n'est plus maintenu depuis de nombreux mois et il est maintenant largement obsolète, je n'apporterai aucune correction.
Mais les contributions des autres développeurs sont les bienvenus

genoue, 2024-01-03:
Merci Doudz pour ton retour.
Aurais-tu une alternative de plugin jeedom à conseiller avec ma zigate ? abeille ?
Merci

doudz, 2024-01-03:
Je conseillerai plutôt une alternative à Jeedom ;) , j'utilise home assistant depuis des années 
Mais au delà de ça, personnellement je n'utilise plus zigate, depuis plusieurs mois j'ai une clé sonoff et le fonctionnement est très stable.

En restant sur jeedom avec zigate, il me semble que le plugin officiel zigbee de jeedom supporte la zigate, en tout cas c'était prévu à un moment

genoue, 2024-01-04:
Bonjour Doudz,
Merci pour ces conseils.
Bonne journée